gumyr/build123d #886: Add a section to the docs on `sort_by`/`filter_by`/`group_by` shape property

gumyr, 2025-01-29:
As a replacement for the `SortBy` functionality of the `ShapeList`, one can use class properties, like:
```py
shape.edges().filter_by(Edge.is_interior)
```
or
```py
from itertools import product

from build123d import Box, ShapeList, Solid, Vertex
from ocp_vscode import ColorMap, show

boxes = ShapeList(
    Box(1, 1, 1).scale(0.75 if (i, j) == (1, 2) else 0.25).translate((i, j, 0))
    for i, j in product(range(-5, 6), repeat=2)
)

# sort by distance to origin
boxes = boxes.sort_by(Vertex().distance)
show(*boxes, colors=ColorMap.listed(len(boxes)))
```
and
```py
# sort by distance to largest box
boxes = boxes.sort_by(boxes.sort_by(Solid.volume).last.distance)
show(*boxes, colors=ColorMap.listed(len(boxes)))
```

[image]

jwagenet, 2025-03-26:
Do you have more examples of the shapelist operators in action? Especially callables since `sort_by(Vertex().distance)` is replaced by sort_by_distance(Vertex()).

gumyr, 2025-03-27:
How about these?
```py
.filter_by(lambda f: len(f.inner_wires()) >= 3) # 3 or more holes
.filter_by(lambda w: all(e.geom_type == GeomType.CIRCLE for e in w.edges())) # All the edges in the wire are circular
.filter_by(lambda w: len(set(e.radius for e in w.edges())) == 1) # wires with a radius of 1
.filter_by(lambda e: cyl_length * 0.4 < e.length < cyl_length * 0.9) # edges of a given length range
.filter_by(lambda f: MotorClass.spline_c1_continuity(f)) # custom methods
.filter_by(lambda f: any(e.geom_type == GeomType.LINE for e in f.edges())) # contains a straight line
.filter_by(lambda f: cylinder_dir.dot(f.bounding_box().size) > 2 * f.radius) # cylinder is longer than diameter
.filter_by(lambda f: len(f.outer_wire().edges().filter_by(GeomType.LINE)) >= 8) # nested filter_by
```

Are these the types of things you're looking for?

jwagenet, 2025-03-27:
Oh wow, yeah these are great! This is way more than I was hoping for. I need to test some of these to fully understand what's going on, but I can make some slightly more fleshed out examples of these.

`filter_by` seems like it can do the most heavy lifting of the 3 operators and `sort_by`/`group_by` are good for fine tuning the filter selections. Do you have any examples where the sorting operators are most useful?

Do you see value in keeping references to `filtered` and `sorted` as suggested ways to mutate ShapeList? I could see an example comparing filtering with `filter_by` with `filtered` and a list comprehension, but I also don't want to confuse people.

Am I missing anything about the ShapePredicate type that it just takes a callable (lambda/func/class method) and returns a bool for the filter?

jwagenet, 2025-03-27:
> Typically I use `sort_by` and `group_by` with Shape properties and not with a lambda although they could be used like this:

Examples not using lambda are great too.

> I was planning for a nice section on topology exploration because it's really quite amazing what can be done

I don't want to step on any ideas you had planned. Is this something that can fit into the section I have put together or is it adjacent? Happy to hear any input you have on the direction I've taken [demo from discord](https://jwagenet.github.io/build123d/html/selectors_operators.html).

> imported models

This is a great idea for some examples. 2d import too.

gumyr, 2025-03-29:
I'm very happy to have the help (just take a look at the growing backlog of issues), thank you.

Here is a property example:
```py
from build123d import *
from ocp_vscode import show

with BuildPart() as tube:
    with BuildSketch():
        Circle(4)
        Circle(3, mode=Mode.SUBTRACT)
    extrude(amount=-10)
    with BuildSketch(tube.faces().sort_by(Axis.Z)[0]):
        Rectangle(10, 10)
    extrude(amount=-2)

    socket_loc = (
        tube.faces().filter_by(Face.is_circular_concave)[0].axis_of_rotation.location
    )
    RigidJoint("socket", joint_location=socket_loc)

    chamfer(tube.faces().filter_by(Face.is_circular_concave).edges(), 0.4)
    fillet(tube.faces().filter_by(Face.is_circular_convex).edges(), 0.4)

show(tube, render_joints=True)
```

[image]

jwagenet, 2025-04-19:
This can be closed with the implementation of #959, unless you want this as a reminder for more examples.

closed 2025-05-15 by gumyr