gumyr/build123d #891: `Axis.intersect` issues

jdegenstein, 2025-01-31:
on current @dev:
```py
from build123d import *
a = Axis.X
b = Axis((0,1),(1,0,0))
c = Axis((0,1),(0,0,1))
d = Axis((0,1),(1,-1,0))
e = Axis((0,1),(1,1,1))
f = Axis((1,1,1),(1,-1))
ra = -Axis.X
print(f"{a.intersect(b)=}")
print(f"{a.intersect(c)=}")
print(f"{a.intersect(d)=}")
print(f"{a.intersect(-d)=}")
print(f"{a.intersect(f)=}")
print(f"{e.intersect(d)=}")
print(f"{e.intersect(-d)=}")
print(f"{ra.intersect(d)=}")
print(f"{ra.intersect(-d)=}")
edge1 = Edge.make_line((3,-7,5),(4,-9,9))
edge2 = Edge.make_line((-7,-8,4),(-4,-7,3))
axis1 = Axis(edge1)
axis2 = Axis(edge2)
print(f"{axis1.intersect(axis2)=}")
```
```
+-----------------------------------------+-----------------------------------------+
|                        current behavior |                        desired behavior |
+-----------------------------------------+-----------------------------------------+
|                     a.intersect(b)=None |                     a.intersect(b)=None |
|          a.intersect(c)=Vector(0, 0, 0) |                     a.intersect(c)=None | #wrong
|          a.intersect(d)=Vector(1, 0, 0) |          a.intersect(d)=Vector(1, 0, 0) | 
|         a.intersect(-d)=Vector(1, 0, 0) |         a.intersect(-d)=Vector(1, 0, 0) |
|          a.intersect(f)=Vector(2, 0, 0) |                     a.intersect(f)=None | #wrong
|          e.intersect(d)=Vector(0, 1, 0) |          e.intersect(d)=Vector(0, 1, 0) |
|         e.intersect(-d)=Vector(0, 1, 0) |         e.intersect(-d)=Vector(0, 1, 0) |
|         ra.intersect(d)=Vector(1, 0, 0) |         ra.intersect(d)=Vector(1, 0, 0) |
|        ra.intersect(-d)=Vector(1, 0, 0) |        ra.intersect(-d)=Vector(1, 0, 0) |
| axis1.intersect(axis2)=Vector(2, -5, 1) | axis1.intersect(axis2)=Vector(2, -5, 1) |
+-----------------------------------------+-----------------------------------------+
```

jdegenstein, 2025-01-31:
Probable solution, including something for unidirectional @snoyer also thanks Ry on discord:
```py
class Axis...
    def intersect...
        ...
                except np.linalg.LinAlgError:
                    return None  # The lines do not intersect

                if abs(d * cross_d1_d2_norm) > TOLERANCE:
                    return None  # The lines do not intersect
                
                # this only allows intersections on the positive axes sides
                # TODO: consider enabling this with an optional parameter
                # if t1 < 0 or t2 < 0:
                #     return None
```

gumyr, 2025-01-31:
I've added a new method to `Axis` - `is_skew` with b8dcad3bcbaf9c788446fbe40189297352db8a9a and added this missing case to intersect.  Are you happy with axis intersect now?

jdegenstein, 2025-02-01:
Looks great, thank you!

closed 2025-02-01 by jdegenstein