gumyr/build123d #898: operations_generic.sweep ignores most parameters when creating faces

shangaslammi, 2025-02-07:
Any Edges (as opposed to Faces) to passed `sweep()` ignore the parameters `multisection`, `is_frenet`, `normal` and ´binormal` so it is impossible to e.g. create a face by sweeping an edge with a fixed normal.

gumyr, 2025-02-07:
One can use the direct api methods to do this:
```py
curve_line = ThreePointArc((2, 0), (0, 0.5), (-2, 0))
curve_face = Face.extrude(curve_line, (0, 0, 4))

line = Line((2, 0, -2), (2, 0, 2))
arc_face = Pos(X=4) * Face.sweep(line, curve_line)
```

[image]

I'm guessing you'd like the `sweep()` operation to support 1D objects?

shangaslammi, 2025-02-12:
Sweep already does support 1D objects, it's just that the additional parameters like `multisection` or `normal` are silently ignored, apparently because `Shell.sweep` does not yet support them: https://github.com/gumyr/build123d/blob/f6f916725e360b96d957fd2d4ba8016f1f4cdbd3/src/build123d/operations_generic.py#L1107

I think those should be straightforward to pass on to OCC's `BRepOffsetAPI_MakePipeShell` function. I can test adding multisection and binormal support also for Shell.sweep and make a PR if it works the way I expect.