gumyr/build123d #900: Plane rotation over Z-axis not working as expected

Niketin, 2025-02-09:
I stumbled upon this line in the documentation while playing with planes and rotating them: 
https://github.com/gumyr/build123d/blob/f22f54af5f4ce5e648d1932d6872cbe572b3d5ad/src/build123d/geometry.py#L2611-L2612

I believe rotation of a plane does not work as described. Rotating plane over Z-axis does not rotate it over its own Z-axis, but over the world's Z-axis. I will demonstrate my observation below.

```python
p1 = Plane.XZ
p2 = p1.rotated((0, 0, 90))
p3 = p2.rotated((0, 0, 90))
p4 = p3.rotated((0, 0, 90))

p2 = p2.move(Location((10, 0, 0)))
p3 = p3.move(Location((20, 0, 0)))
p4 = p4.move(Location((30, 0, 0)))

show(
    location_symbol(p1.location, l=3),
    location_symbol(p2.location, l=3),
    location_symbol(p3.location, l=3),
    location_symbol(p4.location, l=3),
)
```
In the example, I chose Plane.XZ as the base plane. Then I created copies of it, each rotated by 90 degrees from the previous plane, around the Z-axis. Then, the planes are moved so that they are not on top of each other, for easier visualization.
[image]
As can be seen from the picture, The plane is rotated around its Y-axis, which is parallel to the world's Z-axis.

Is it a bug? In case it is not a bug and it works as intended, I have a follow-up question: how can I rotate a plane around its own Z-axis?

gumyr, 2025-02-09:
The rotations are global.  You want to set/change the Plane's x_dir.  
```py
pln0 = Plane.XY
pln1 = Plane(origin=pln0.origin, x_dir=(0, 1, 0), z_dir=pln0.z_dir)
```

Niketin, 2025-02-20:
@gumyr I see. Then the part of the documentation referred above doesn't really make sense to me. The statement seems false in the context of rotating a plane using this method.

commit 2168fd05 (2025-03-27): Plane.rotated(): remove incorrect note about z rotation to resolve #900
closed 2025-03-27 by gumyr