gumyr/build123d #907: Shape.compute_mass() for Compounds not reliable.

slobberingant, 2025-02-20:
In Compounds with parts of sufficient complexity, the Shape.compute_mass() function returns an incorrect value.
Sample code below.

```
b1 = Cylinder(100,1000) - Cylinder(80, 1000) - Cylinder(90, 200).move(Rot(90)) 
b2 = Box(100,100,100)
b2.move(Pos(1000,1000,1000)) # way out of the way so no intersection.

c1 = Compound(children=[b1,b2])
print('Comparison between volume and compute_mass in Compound with 2 parts.')
print(f'b1 : {b1.volume == Shape.compute_mass(b1)}') # True
print(f'b2 : {b2.volume == Shape.compute_mass(b2)}') # True
print(f'c1 : {c1.volume == Shape.compute_mass(b1) + Shape.compute_mass(b2)}') # True
print(f'c1 : {c1.volume == Shape.compute_mass(c1)}') # False

c2 = Compound(children=[b1,])
print('Comparison between volume and compute_mass for single entity Compound.')
print(f'b1 : {b1.volume == Shape.compute_mass(b1)}') # True
print(f'b2 : {b2.volume == Shape.compute_mass(b2)}') # True
print(f'c1 : {c2.volume == Shape.compute_mass(c2)}') # True
print(f'c1 : {c2.volume == Shape.compute_mass(b1)}') # True
```

This only occurs when calling Shape.compute_mass() directly on a Compound. Other sections of build123d code, seem to iterate over solids or parts within a Compound and return correct values. Unfortunately the iterative process does not consider parts that intersect and may over estimate the volume if two parts occupy the same space.

Considerations:
- Any solution should allow for a density value to be easily added so that a part weight can be returned.
- Intersections should be clearly handled. Either with a warning or clear documentation.
- There should be a function that will return the total volume of all parts in a Compound rather than the first level only.

gumyr, 2025-03-06:
Here are a few things that need to be considered as part of this change:

- Enabling density depends on issue https://github.com/gumyr/build123d/issues/598 where a material system is discussed. Standard in this area are lacking making the choice of solution more difficult.
- Calculating the intersection of all parts in an assembly could be slow with a large assembly so it isn't clear if all users would design this as a separate check could be made to determine intersections.
- The solution must be able to handle an Solids in an arbitrarily complex assembly.

Discussion very welcome.

slobberingant, 2026-09-17:
compute_mass() now considers density so these two functions are no longer equivalent.

closed 2026-09-17 by slobberingant