gumyr/build123d #915: Vertical dimension line label scoring

slobberingant, 2025-02-25:
Vertical dimension lines do not position their label correctly inside even though there is sufficient space.

```
draft = Draft()
line = DimensionLine([(0,0), (0, 100)], draft)
```

[image]

slobberingant, 2025-03-28:
DimensionLine determines where to put the label based on a scoring system which is affected by any common area (overlap) between the label and associated lines. The existing code:

```score = (d_line.area - 10 * common_area) / bbox_size.X```

The bbox_size.X only considers the size in the X direction which will always be incorrect for a vertical dimension.
An option is to change the code to:

```score = (d_line.area - 10 * common_area) / max(bbox_size.X, bbox_size.Y)```

This will get the largest bounding box.
The issue with this is that when you do angled dimensions you may run into issues. Is there any way to determine the size of an angled bounding box across it's faces?

gumyr, 2025-03-28:
Yes - `bbox.diagonal`.  Suggestion on how to improve the "scoring system" would be appreciated.

Drawing dimension lines is a place where an AI would be really helpful - "for this sketch, create the code to generate a fully dimensioned technical drawing following appropriate standards."

closed 2025-05-30 by gumyr