gumyr/build123d #916: Basic cone generates bugged geometry

MatthiasJ1, 2025-02-27:
```py
Cone(10, 0, 20)
```

Gives:
[image]

gumyr, 2025-03-04:
It seems like the problem may be in display and not in the object itself.  If I do:
```py
sections = [section(c, Plane.XY.offset(height)) for height in [8, 9, 9.5, 9.75]]

calculated_volume = pi * 10**2 * 20 / 3

print(calculated_volume, c.volume, abs(calculated_volume - c.
```
```
2094.3951023931954 2094.395102393195 4.547473508864641e-13
```
the volumes are the same and the sections look as follows:

[image]

Any suggestion on how to isolate this to a build123d vs display problem?

snoyer, 2025-03-05:
> Any suggestion on how to isolate this to a build123d vs display problem?

[F3D](http://f3d.app) can open both `.brep` and `.vtp` files so you could compare the OCCT dump and the VTK conversion

```py
from pathlib import Path
from build123d import Cone, export_brep
from build123d.vtk_tools import to_vtkpoly_string

cone = Cone(10, 0, 20)
export_brep(cone, "/tmp/cone.brep")
Path("/tmp/cone.vtp").write_text(to_vtkpoly_string(cone))
```

Triangulation looks ok on both `.brep` and `.vtp` files but they both show the same kind of issue near the tip which looks like it's due to the normals (not really sure what's up with that tho, F3D uses face normals from OCCT for the `.brep`, and the `.vtp` has normals computed by VTK on export via `build123d`).

On the `build123d` side, changing the `FeatureAngle` param to `90` seems to fix the tip artifact but the normals aren't smoothed as nicely everywhere
https://github.com/gumyr/build123d/blob/e9c6b93cb70d7c8eab149330254ff7acab6e12ca/src/build123d/vtk_tools.py#L116

gumyr, 2025-03-05:
When I "porcupine" the cone the normals all look good well into the zone that would have trouble:

[image]

Does OCCT have some special logic when tessellating a cone to handle the tip?  Maybe we're seeing an artifact of that.

snoyer, 2025-03-05:
Looks like the cone itself is fine but there's something going on when exporting through VTK.
For example, exporting to `.vtp` after changing the `n_filter.SetFeatureAngle(360)` line as mentioned in my earlier message I see (visualizing the Y-component of the normals because F3D doesn't do hedgehogs yet):

[image]
[image]
[image]
[image]
[image]

with `360` (current default) and `180` there is a discontinuity at the tip and the reflection/highlight "bends" the same as in the original screenshot. With lower values this discontinuity goes away but the rest of the normals aren't as smooth.