gumyr/build123d #922: Racey import/export tests with pytest-xdist

hcsch, 2025-03-01:
Because pytest-xdist is listed in the `pyproject.toml` I used pytest with xdist. This however leads to some tests like e.g. `TestMeshProperties.test_properties` in `test_mesher.py` failing depending on the execution timing of the tests and what other tests are running in parallel. This is because multiple tests are racing to write/read from the same file.

These issues can be avoided by using a tempdir/tempfile per test, instead of a fixed file name in the working directory like e.g. "test.3mf". pytest has a fixture for that ([`tmp_path`](https://docs.pytest.org/en/stable/how-to/tmp_path.html)), however I wasn't able to get that to work within the `unittest.TestCase` classes. If you'd like I can open a PR which adds a `TmpPathTestCase` to the affected test files, which sets `self.tmp_path` to a temporary directory specific to each test that gets cleaned-up afterwards. If you want to generally restructure tests more towards pytest, using the `tmp_path` fixture would probably be cleaner in the long run though.

gumyr, 2025-03-06:
I think you must have noticed this problem during the short window that this new xdist functionality was being introduced as your suggestions have already been implemented in the tests.  Please take a look and let me know if you still think there are issues with any of the tests.

hcsch, 2025-03-07:
Ah, sorry in that case. I should have had a closer look at the `dev` branch as well and not just the latest release that I was packaging in nix. I'll try building a newer version some time soon and will close the issue when I've confirmed that this has been resolved already :)

jdegenstein, 2025-11-20:
@hcsch can you confirm this issue can be closed?