gumyr/build123d #924: Separate out dependencies that are not necessary for 3D object creation

BlueDrink9, 2025-03-01:
To simplify the installation of build123D for use as a build library, rather than as a design process tool, it would be good to explicitly separate out those dependencies that are only for interactive use. This would then be specified as a special argument to the pip installation, e.g. `pip install "build123d[interactive]"`. This would have the downside of slightly complicating installation for users that need that feature (but this should be easy to remedy with an include guard that prints out the required installation steps for the pretty printing action).

The crux of this issue really is ipython, which is used for pretty printing in build123d but breaks Google Collab when updated (explicitly - they only support one version). I'd like to use Collab to allow people to easily run and customize objects with [snakeskin](https://GitHub.com/BlueDrink9/keeb_snakeskin) without installing it on their own machine.

As an alternative, would you welcome a PR to remove the ipython dependency in favour of some simpler dependency?

gumyr, 2025-03-02:
There is a larger effort to separate all of the display functionality out of build123d so in general there is alignment to this request.  `jupyter_tools.py` would still need to be handled.

With respect to the `IPython.pretty` functionality, how about replacing it with the follows?
```py
from build123d import *
from build123d import GroupBy
from typing_extensions import Iterable

v = Box(1, 1, 1).vertices().group_by(Axis.Z)
print(v)

def GroupBy__repr__(self):
    return self._custom_pretty()

def GroupBy_custom_pretty(self, level=0):
    """
    Recursively build a bracketed, multi-line representation.
    """
    indent = "  " * level
    lines = []
    lines.append(indent + "[")  # opening bracket

    for idx, item in enumerate(self):
        if idx > 0:
            lines[-1] += ","  # append comma to previous line
        if not isinstance(item, (Vector, Vertex)) and isinstance(item, Iterable):
            lines.append(GroupBy_custom_pretty(item, level + 1))
        else:
            lines.append(indent + "  " + str(item))

    lines.append(indent + "]")  # closing bracket
    return "\n".join(lines)

GroupBy.__repr__ = GroupBy__repr__
GroupBy.__str__ = GroupBy__repr__
GroupBy._custom_pretty = GroupBy_custom_pretty
print(v)
```
```
[[Vertex(-0.5, -0.5, -0.5),
  Vertex(-0.5, 0.5, -0.5),
  Vertex(0.5, -0.5, -0.5),
  Vertex(0.5, 0.5, -0.5)],
 [Vertex(-0.5, -0.5, 0.5),
  Vertex(-0.5, 0.5, 0.5),
  Vertex(0.5, -0.5, 0.5),
  Vertex(0.5, 0.5, 0.5)]]
[
  [
    Vertex(-0.5, -0.5, -0.5),
    Vertex(-0.5, 0.5, -0.5),
    Vertex(0.5, -0.5, -0.5),
    Vertex(0.5, 0.5, -0.5)
  ],
  [
    Vertex(-0.5, -0.5, 0.5),
    Vertex(-0.5, 0.5, 0.5),
    Vertex(0.5, -0.5, 0.5),
    Vertex(0.5, 0.5, 0.5)
  ]
]
```
The output is quite similar and avoids the use of IPython.

BlueDrink9, 2025-03-02:
Fantastic! That seems good to me

jdegenstein, 2025-03-02:
@BlueDrink9 what are you using to visualize your models? Are you using the inline jupyter viewer? If so, that is the primary reason that build123d depends on ipython. 

Either way I support the idea that jupyter_tools and ipython could be separated out as deps of build123d. We could even consider adding another package to this repository for jupyter_tools that is an optional dep of build123d as discussed above.

We currently don't have a good replacement for jupyter_tools either, but something like [jupyter-cadquery](https://github.com/bernhard-42/jupyter-cadquery) might be ideal to remove our need for VTK as well.

BlueDrink9, 2025-03-02:
Good question, I hadn't actually gotten that far. I was going to just go hunting for options, assuming that .stl previewers weren't uncommon and that they would be happy with the collab ipython version