gumyr/build123d #925: Shape.edges() fails to return majority of edges

MatthiasJ1, 2025-03-03:
```py
s = GridLocations(10,10,5,5) * Rectangle(10,10)
s = Curve() + [f.wire() for f in s]
```

creates the following geometry:

[image]

```py
e = s.edges()
```

returns the following edges:

[image]

gumyr, 2025-03-04:
The problem here is with `s = Curve() + [f.wire() for f in s]` as this attempts to create a `Wire` given all of the inputs are `Wires` but that doesn't work as a single `Wire` can't be created from these objects.  There needs to be more logic in this to ensure a `Wire` can actually be created, if not a `Compound` needs to be created which isn't easy given the structure of the code.

The quick fix is to do: `s3 = Compound() + [f.wire() for f in s]`.

jdegenstein, 2025-03-04:
What is the output of show_topology?

MatthiasJ1, 2025-03-04:
show_topology gives all the edges. `show` has to be getting them from somewhere so I would assume they are there.

snoyer, 2025-03-05:
Most likely a `BRepTools_WireExplorer` vs `TopoDS_Iterator` question, see #864

gumyr, 2025-03-05:
I should have clarified - `show_topology` does show the edges but we're kinda in a weird state where this isn't really a good `Wire` and other wire methods are going to have trouble.  As a solution how about the `__add__` method of `Curve`/`Compound` keeps the object as a `Curve`/`Compound` and never attempts to create a `Wire`? The goal of attempting to create a `Wire` was to make it easier to use the other "line like" methods but this example shows the danger in that.

gumyr, 2026-10-03:
Fixed with PR #1479

closed 2026-10-03 by gumyr