gumyr/build123d #939: Offset doesn't work on shifted Sphere

fxdave, 2025-03-15:
## Works:
```py
import build123d as b
import ocp_vscode as ocp

with b.BuildPart() as test:
    b.Sphere(100)
    b.offset(amount=2)

ocp.show_object(test)
```
## Dosen't work:
```py
import build123d as b
import ocp_vscode as ocp

with b.BuildPart() as test:
    with b.Locations((0, 0, 80)):
        b.Sphere(100)
    b.offset(amount=2)

ocp.show_object(test)
```

RuntimeError: offset Error, an alternative kind may resolve this error
But Kind.INTERSECTION doesn't resolve it either.

fxdave, 2025-03-16:
Another simple object that fails:

```py
import build123d as b
import ocp_vscode as ocp

ocp.reset_show()

e1 = [130, 93]
backRadius = 2

with b.BuildPart() as ellipse1:
    with b.BuildSketch():
        b.Ellipse(e1[0] / 2, e1[1] / 2)
        b.split(bisect_by=b.Plane.XZ)
    b.revolve(axis=b.Axis.X, revolution_arc=180)

with b.BuildPart() as ellipse2:
    with b.BuildSketch():
        b.Ellipse(e1[1] / 2, e1[0] / 2)
        b.split(bisect_by=b.Plane.XZ)
    b.revolve(axis=b.Axis.X, revolution_arc=360)

with b.BuildPart() as ellipse3:
    with b.BuildSketch():
        b.Ellipse(e1[1] * 0.4, e1[0] * 0.4)
        b.split(bisect_by=b.Plane.XZ)
    b.revolve(axis=b.Axis.X, revolution_arc=360)

with b.BuildPart() as body:
    b.add(ellipse1)
    with b.Locations((65, -30, -e1[1] / 2)):
        b.add(ellipse2, mode=b.Mode.SUBTRACT)

    with b.Locations((-65, 30.0, -e1[1] / 2)):
        b.add(ellipse3, mode=b.Mode.SUBTRACT)

    b.fillet(body.edges(), radius=backRadius)
    b.offset(amount=2) # !!! Standard Construction Error

ocp.show_object(body, name="body")
```

gumyr, 2025-03-16:
What you are seeing are OpenCascade (OCCT, the CAD kernel that build123d uses) errors.  It just can't do an offset for some shapes.  See: https://build123d.readthedocs.io/en/latest/tips.html#can-t-get-there-from-here

There is very little to nothing that can be done with fix these problems at the build123d level.

fxdave, 2025-03-16:
@gumyr Thanks for the quick response! Is there a possible alternative to grow an already built shape by its normals? I'm working on a mold generator for climbing holds. I'm mostly done but I depend on the offset command unfortunately. It's so simple operation in polygon modelling, although I understand CAD is different and complex.

gumyr, 2025-03-16:
If your objects are parametric you should be able to change the parameters and create a larger/smaller version of the shape.  Alternatively, as a rough approximation you could scale the shape.

fxdave, 2025-03-16:
That might do. Thanks again!

closed 2025-03-16 by fxdave