gumyr/build123d #940: Docs: Add example of a spiral ramp

KilowattSynthesis, 2025-03-17:
I can't figure out how to draw a spiral ramp. I've looked at building off the spring example in #759, but still can't figure it out. 

Would appreciate it if someone could help draw an example, and then later add it to the docs

gumyr, 2025-03-17:
Is this what you are looking for?
```py
profile = make_face(
    offset(Polyline((-0.5, 0.2), (-0.5, 0), (0.5, 0), (0.5, 0.2)), 0.1, side=Side.RIGHT)
)
ramp_path = Helix(3, 6, 10)
ramp = sweep(ramp_path.location_at(0) * profile, ramp_path, is_frenet=True)
```

[image]

KilowattSynthesis, 2025-03-17:
Yes, that's exactly it, thank you! Was having trouble getting the base of the ramp to be flat. Thanks a lot!

KilowattSynthesis, 2025-03-17:
I'm trying to fill in the part below the ramp to the XY plane. Any idea why this `extrude(..., until=ramp)` isn't working?

```python
from build123d import *
import build123d as bd
import ocp_vscode

profile = make_face(
    offset(Polyline((-0.5, 0.2), (-0.5, 0), (0.5, 0), (0.5, 0.2)), 0.1, side=Side.RIGHT)
)
ramp_path = Helix(3, 6, 10)
ramp = sweep(
    ramp_path.location_at(0) * profile,
    ramp_path,
    is_frenet=True,
)

extrude_part_base = bd.Sketch(bd.Circle(9.8) - bd.Circle(9.2))

# Following part fails.
extrude_part = bd.extrude(
    extrude_part_base,
    until=ramp,
)

ocp_vscode.show_all()
```

gumyr, 2025-03-17:
The API of `extrude` isn't as you're using it - something like:
```py
extrude_part = bd.extrude(extrude_part_base, until=Until.NEXT, target=ramp)
```
However, the implementation isn't robust enough to handle a helical object like that (spanning 720 degrees) , especially when the `extrude_part_base` isn't even completely contained within the ramp (its radius is smaller than the ramp minimal radius).  I don't even know what that would look like.

To create a support for the ramp, limit the ramp to 360 degrees (or create another object for this) and extrude the bottom face downward then intersect the result with a hollow cylinder.

KilowattSynthesis, 2025-03-17:
Oops, thanks a lot!

This worked well:
```
from build123d import *
import build123d as bd
import ocp_vscode

profile = make_face(
    offset(Polyline((-0.5, 0.2), (-0.5, 0), (0.5, 0), (0.5, 0.2)), 0.1, side=Side.RIGHT)
)
ramp_path = Helix(3, 3 * 0.75, 10)
ramp = sweep(
    ramp_path.location_at(0) * profile,
    ramp_path,
    is_frenet=True,
).translate((0, 0, 3))

extrude_part_base = bd.Sketch(
    bd.Circle(10.4)
    - bd.Circle(9.6)
    - bd.Rectangle(100, 100, align=(bd.Align.MIN, bd.Align.MAX))
)

extrude_part = bd.extrude(
    extrude_part_base,
    until=bd.Until.NEXT,
    target=ramp,
)

ocp_vscode.show_all()
```

[image]