gumyr/build123d #941: Feature Request: Add a `Polyhedron` object, which is constructed from a list of 3d points

KilowattSynthesis, 2025-03-17:


gumyr, 2025-03-17:
There is an example doing this here: https://build123d.readthedocs.io/en/latest/examples_1.html#platonic-solids 

Basically: `solid_obj = Solid(Shell(list_of_faces))`

What is the application for doing this?  It seems quite specialized.

KilowattSynthesis, 2025-03-17:
I did read that example. 

Instead of a list of faces, I'm hoping to work directly with a list of points (for the very-obscure irregular shapes). Making a list of faces from a list of points is tough.

Alternatively, a 3d convex hull would also solve this problem.

gumyr, 2025-03-17:
The code in that example can be used to make random rocks, you just need to provide the list of points.
```py
from build123d import *
from ocp_vscode import show
import numpy as np
from scipy.spatial import ConvexHull

def generate_rock_points(n_points=100, noise_strength=0.2):
    """Generate a set of 3D points resembling a natural rock shape."""
    # Generate spherical points
    phi = np.random.uniform(0, np.pi, n_points)  # Polar angles
    theta = np.random.uniform(0, 2 * np.pi, n_points)  # Azimuthal angles

    # Convert spherical coordinates to Cartesian (x, y, z)
    x = np.sin(phi) * np.cos(theta)
    y = np.sin(phi) * np.sin(theta)
    z = np.cos(phi)

    # Apply random Gaussian noise for roughness
    x += np.random.normal(0, noise_strength, n_points)
    y += np.random.normal(0, noise_strength, n_points)
    z += np.random.normal(0, noise_strength, n_points)

    points = np.vstack((x, y, z)).T  # Combine into a (n_points, 3) array
    return points

rock_vertices = generate_rock_points()

# Create a convex hull from the vertices
hull = ConvexHull(rock_vertices).simplices.tolist()

# Create faces from the vertex indices
rock_faces = []
for face_vertex_indices in hull:
    corner_vertices = [rock_vertices[i] for i in face_vertex_indices]
    rock_faces.append(Face(Wire.make_polygon(corner_vertices)))

# Create the solid from the Faces
rock = Solid(Shell(rock_faces)).clean()

show(rock)
```

[image]

I guess those few lines of code could be wrapped up into a new object - wouldn't make it easier to use though.

KilowattSynthesis, 2025-03-18:
May be obvious to you, but I want to assure you that, even as someone who uses build123d regularly, the following code is still black magic to me (and thus likely to much of the community):

```python

# Create a convex hull from the vertices
hull = ConvexHull(rock_vertices).simplices.tolist()

# Create faces from the vertex indices
rock_faces = []
for face_vertex_indices in hull:
    corner_vertices = [rock_vertices[i] for i in face_vertex_indices]
    rock_faces.append(Face(Wire.make_polygon(corner_vertices)))

# Create the solid from the Faces
rock = Solid(Shell(rock_faces)).clean()
```

`ConvexHull` doesn't even appear in the documentation (other than that specific example). `.simplices` doesn't appear to be documented anywhere. Using indexes from one place into another list is something I never would have thought to consider.

> I guess those few lines of code could be wrapped up into a new object - wouldn't make it easier to use though.

Why don't you think it "wouldn't make it easier to use"? Seems pretty obvious to me that it _would_ make it easier to use, especially given the fact that the example you've laid out here relies on undocumented features (or perhaps features only described by a single example).

jdegenstein, 2025-03-18:
> ConvexHull doesn't even appear in the documentation

https://docs.scipy.org/doc/scipy/reference/generated/scipy.spatial.ConvexHull.html

KilowattSynthesis, 2025-03-18:
Ugh the `import *` got me again. Because everything in these examples magically comes from build123d, I assumed this one did as well. 

Anyway, relying on a random feature from scipy seems sorta against the spirit of build123d?

gumyr, 2025-03-18:
Adding a `Polyhedron` object isn't a bad idea I'm just showing you how you can achieve that now without waiting. 

One of the most powerful aspects of build123d is that this is a one of 616,406 Python projects (on pypi) which allows users to use a vast collection of open-source content in their projects so "relying on a random feature from scipy" is 100%  aligned with the spirit of build123d.