gumyr/build123d #943: Fillet Not Handling Slicing Correctly

barrettluke, 2025-03-19:
Description
When applying fillets using Build123d, methods 1-4 fillet all four Z-axis edges as expected but unexpectedly fillet two edges along the X-axis and two along the Y-axis at the bottom. However, methods 5 and 6 behave as expected when slicing edges.

Steps to Reproduce
	1.	Use the following script with Build123d and ocp_vscode:
```python
from build123d import *
from ocp_vscode import show

# Dimensions
length, width, thickness = 120.0, 110.0, 30.0
fillet_radius = 3.0
with BuildPart() as ex1:
    # Create the box
    Box(length, width, thickness)

    # Method 1 
    # fillet(ex1.edges().filter_by(Axis.Z), radius=fillet_radius)
    # fillet(ex1.edges().filter_by(Axis.X)[-1], radius=fillet_radius)
    #-----------------------------------
    # Method 2
    # fillet(ex1.edges().filter_by(Axis.Z), radius=fillet_radius)
    # test = ex1.edges().filter_by(Axis.X)[-1]
    # fillet(test, radius=fillet_radius)
    #-----------------------------------
    # Method 3
    # fillet(ex1.edges().filter_by(Axis.Z), radius=fillet_radius)
    # fillet(test := ex1.edges().filter_by(Axis.X)[2], radius=fillet_radius)
    #-----------------------------------
    # Method 4
    # fillet(ex1.edges().filter_by(Axis.Z), radius=fillet_radius)
    # fillet(test := ex1.edges().filter_by(Axis.X)[-1], radius=fillet_radius)
    #-----------------------------------
    # Method 5 (Expected Behavior)
    # axis_z = ex1.edges().filter_by(Axis.Z) + ex1.edges().filter_by(Axis.X)[::2] + ex1.edges().filter_by(Axis.Y)[1::2]
    # fillet(axis_z, radius=fillet_radius)
    #-----------------------------------
    # Method 6 (Expected Behavior)
    # b_half = ex1.edges().filter_by(Axis.Z) + [ex1.edges().filter_by(Axis.X)[-1]]
    # fillet(b_half, radius=fillet_radius)

# Show the result
show(ex1.part)
```
	2.	Uncomment methods 1-4 one at a time and observe that they fillet the four Z-axis edges as expected, but also fillet unexpected edges on the X and Y axes at the bottom.
	3.	Uncomment methods 5 and 6 to see that they behave as expected.

Expected Behavior
Fillets should only be applied to the specified edges without modifying additional edges along the X and Y axes unless explicitly included.

Observed Behavior
	•	Methods 1-4 fillet unintended edges on the bottom of the shape.
	•	Methods 5 and 6 correctly apply fillets without affecting unwanted edges.

Environment
	•	Build123d Version: 0.9.2.dev46+g32a1ea1
	•	Python Version: 3.10
	•	OS: MacOS Sequoia Version 15.2

Additional Context
It seems like the filter_by method isn’t correctly handling edge selection, leading to unexpected filleting behavior. Could this be an issue with how edges are indexed or stored in Build123d?

gumyr, 2025-03-20:
Thanks for reporting this problem.  The bug is within the OpenCascade CAD kernel (OCCT) so there is little that can be done about it.  The following screenshots show the edge that should be filleted and the resulting part where all four of the bottom edges are filleted. The build123d `filter_by` is working as expected.  One can show the results of the filters in the show function:

```py
    # Method 1
    e1 = ex1.edges().filter_by(Axis.Z)
    fillet(e1, radius=fillet_radius)
    e2 = ex1.edges().filter_by(Axis.X)[-1]
    fillet(e2, radius=fillet_radius)

show(ex1.part, e1, e2)
```
[image]

The order of operations doesn't seem to change the behaviour (by the way, one shouldn't index shape lists unless sorting them first, the order is essentially random):
```py
    # Method 7
    e2 = ex1.edges().filter_by(Axis.X).group_by(Axis.Z)[-1]
    fillet(e2, radius=fillet_radius)
    e1 = ex1.edges().filter_by(Axis.Z)
    fillet(e1, radius=fillet_radius)
```
[image]

However, if one combines the edges into a single fillet operation the behaviour is as expected:
```py
    # Method 8
    e1 = ex1.edges().filter_by(Axis.Z)
    e2 = ex1.edges().filter_by(Axis.X).group_by(Axis.Z)[-1]
    fillet(e1 + e2, radius=fillet_radius)
```
[image]

There seems to be more activity within the OCCT community lately.  Hopefully some of these problems will be fixed soon.

gumyr, 2026-10-03:
There is note in the docs now that fillets propagate and given this is an OCCT behaviour there isn't all that much more that can be done.

closed 2026-10-03 by gumyr