gumyr/build123d #946: Compile to wasm

mordv, 2025-03-22:
There is a closed issue: https://github.com/CadQuery/cadquery/issues/1385 claiming that
> we would need to compile OCP to wasm and that's not easy task

There is examples of opencascade running in browser, namely [replicad](https://github.com/sgenoud/replicad). It's not as feature rich, but has much better DX. 

I think the benefits of the ability to run the thing in browser are evident. Is that possible?

gumyr, 2025-03-22:
By "running in browser" do you mean on some other computer (i.e. not locally)?  If so, who hosts this computer and how is security provided such that users can enter Python code (and import other packages) without the machine mining bitcoin?

jdegenstein, 2025-03-22:
build123d also depends on OCP, and I agree with the prior comment that compiling to WASM is not trivial. replicad instead depends on OpenCascade.js which hasn't had a commit in about 2 years (possibly a dead project).

I agree that making things easier to run is a very important goal, and in some respects installing build123d and a viewer is very unapproachable for a python newcomer.

So in summary, I think this is a nice goal but is a huge project unto itself that would require more time than I am willing to devote to it.

mordv, 2025-03-22:
@jdegenstein thanks for clarification! The issue I mentioned is 2y old, I was hoping something has changed in regards of compiling complexity. Can you elaborate where is it come from? Is that a compiling a wasm module or some after chore? 

> replicad instead depends on OpenCascade.js

They not just depend on it (install it) but instead make some [custom](https://github.com/sgenoud/replicad/tree/main/packages/replicad-opencascadejs) build. 

> in some respects installing build123d and a viewer is very unapproachable for a python newcomer

It's ok for newcomers. After all it's python and they should get used to installing stuff (which often breaks)

The real benefit would be for the end user(the one who gets the model) and the developers of tools - look at [example from replicad](https://studio.replicad.xyz/share/https%3A%2F%2Fraw.githubusercontent.com%2Fsgenoud%2Freplicad%2Fmain%2Fpackages%2Freplicad-docs%2Fexamples%2FsimpleVase.js) - it's a fully configurable, downloadable model, generated right in the browser (even params are taken from ulr). The thingiverse has customizer IFAIK it's based on OpenSCAD.

@gumyr 
The 2 tendencies are clear - everything moves to the browser and (I much as I hate it) - python here to stay. [Pyodide](https://github.com/pyodide/pyodide) can run numerous of libs and optionally provide them a html canvas to draw on. So imagine something like [py.cafe](https://py.cafe/) but for build123d - that'd be cool!

Given all that, the benefits go far beyond better newcomer experience - it spreads adoption and distribution significantly. 

Thanks for your work and time!

Jojain, 2025-03-23:
The real issue is that pywrap that is used to build OCP is pretty much undocumented and a veritable pain to understand. 
For the benefits of it we all agree that it would be great so questions about that.

If you have experience with build system you could try to take a look. We documented our journey here https://discord.com/channels/964330484911972403/1079454322221396008

jsmnbom, 2025-03-24:
I did some experimentation with this with the eventual goal of getting build123d running in something like jupyterlite.
I gave up along the way - especially since I realized that the performance simply wouldn't make much sense. As far as I know pyodide runs 3-4 times slower than native, and webassembly itself is often 2 times slower. The opencascade kernel is not exactly fast in the first place.

But in case anyone is interested here's some info. I used the [run_docker](https://github.com/pyodide/pyodide/blob/main/run_docker) form pyodide to get a stable environement for tests. Then after cloning [OCCT](https://github.com/Open-Cascade-SAS/OCCT) we can build it using something like:
```sh
cd OCCT
emcmake cmake -G Ninja . \
  -DUSE_FREETYPE=OFF \
  -DUSE_TK=OFF\
  -DBUILD_LIBRARY_TYPE=Static \
  -DBUILD_MODULE_ApplicationFramework=OFF \
  -DBUILD_MODULE_DETools=OFF \
  -DBUILD_MODULE_DataExchange=OFF \
  -DBUILD_MODULE_Draw=OFF \
  -DBUILD_MODULE_FoundationClasses=ON \
  -DBUILD_MODULE_ModelingAlgorithms=ON \
  -DBUILD_MODULE_ModelingData=ON \
  -DBUILD_MODULE_Visualization=ON
cmake --build .
cmake --build . -- install
```

OCP is unfortunately much harder to compile, and that is where I eventually gave up. But I did get pretty far by adding Emscripten as a new platform whereever it makes sense. Then running the first step of pywrap with `pywrap -n 7 -l /usr/lib/llvm-19/lib/libclang.so parse ocp.toml tmp.pkl Emscripten`. This process is extremely slow on my machine, and I ran out of patience. I suspect that to properly build it, the build process needs to exclude more files than it currently does, but it is too complicated for me to understand.

closed 2025-04-13 by gumyr