gumyr/build123d #949: Should use of SortBy enums or properties work gracefully if they don't exist?

jdegenstein, 2025-03-25:
```py
a = Cylinder(1,1)
a.edges().filter_by(GeomType.CIRCLE).sort_by(Edge.radius) # works
a.edges().filter_by(GeomType.CIRCLE).sort_by(SortBy.RADIUS) # works
a.edges().sort_by(Edge.radius) # fails "ValueError: Shape could not be reduced to a circle"
a.edges().sort_by(SortBy.RADIUS) # fails "ValueError: Shape could not be reduced to a circle"
```

Should the latter two cases return the same as the former two cases? (i.e. an implicit filter)

jwagenet, 2025-03-25:
To expand my initial thoughts:

`part.edges().filter_by(GeomType.CIRCLE).group_by(Edge.radius)` requires the `GeomType` filter to avoid a ValueError kickback from OCP for an incompatible `GeomType` in a mixed-type list of edges.  `ShapeList` methods like `group_by(Edge.radius)` or `filter_by(lambda e: e.radius == 8)` could handle the errors so the `GeomType` filter isn't necessary **or** the properties themselves dont throw the error.

If handling within the properties, returning `None` seems safest.

If handling in operators, filter can could treat error cases as the False case. 
For sort/group its a little more complicated as returning a 'zero' value eg `Line().radius == 0 `really right in this case. Returning None wouldn't play well with sort, unless error/`None` values are implicitly filtered. `None` can work as a key in `GroupBy`, but it would need to be excluded from list slicing.

gumyr, 2025-03-27:
I'm not sure that making `sort_by` into an implicit `filter_by` is very intuitive (and it probably breaks some of the Python guidelines). I don't think any sort should have the property of changing the length of the list.