gumyr/build123d #951: `operations_generic.project` has some rough edges.

jwagenet, 2025-03-26:
Project doesn't like projecting a cylindrical face:
````python
with BuildPart() as part:
    Box(1,1,1)
    Cylinder(.25, 1, mode=Mode.SUBTRACT)

p = part.part.moved(Location(Vector(), (10,10,10)))

with BuildSketch() as sketch:
    # doesnt work
    project(p.faces())
    project(p.faces()[6])

    # works
    project(p.faces()[0:6])
````

Projecting faces/verts on BuildLine gives a rasies `ValueError: Points and faces can only be projected in PRIVATE mode`
````python
with BuildLine() as line:
    project(p.vertices())
    project(p.faces(), mode=Mode.PRIVATE)
````
But private mode yields a different OCP ValueError. Which means the docstring is inaccurate about working for 0D objects (verts)

Trying to project edges in BuildSketch gives `ValueError: Edges, wires and points can only be projected in PRIVATE mode`, but using private gives `ValueError: Points and faces can only be projected in PRIVATE mode`!

````python
with BuildSketch() as sketch:
    project(p)
    project(p.solids())
````
Return an empty object with no error!

### Auto object correction
Outside of the special private mode case, I think using a higher order object in the wrong context should lead to the object being corrected to the correct object type and working anyway. eg in BuildLine `Face` and `Solid` objects are corrected to `Edge` list and projected.

gumyr, 2025-03-27:
Projection is probably the most broken part of build123d.  I made some progress on totally designing it (see #563) but haven't completed the work.  With respect to the `project` operation I think it's implementation is confusing and it attempts to do too much.  I would be very interested in your opinion on how it could be redesigned to make it intuitive to use.

jwagenet, 2025-03-27:
Sorry, I did a cursory search and didn't see the other issue. I'll have to read the history there and play around with the current implementation more.

closed 2026-11-03 by gumyr