gumyr/build123d #953: copy.copy on a Shape is surprisingly slow

jsmnbom, 2025-03-28:
Using `copy.copy` on a `Shape` to create a shallow copy of the shape is a lot slower than I would have imagined from the documentation at https://build123d.readthedocs.io/en/latest/assemblies.html which implies that it is very very fast.

Example:
```sh
$ uv run -m timeit -s "import build123d,copy;screw=build123d.import_step('M6-1x12-countersunk-screw.step')" "copy.deepcopy(screw)"
200 loops, best of 5: 1.61 msec per loop

$ uv run -m timeit -s "import build123d,copy;screw=build123d.import_step('M6-1x12-countersunk-screw.step')" "copy.copy(screw)"
200 loops, best of 5: 1.54 msec per loop
```

This seems to imply that is not actually faster to make a shallow copy rather than a deep one. When actually outputting to a .step as in the documentation example, it is still MUCH faster however, so that is true, but I still found this behavior to be weird.

Looking into the code, it seems that `__copy__` actually calls `deepcopy`:
```python
class Shape(NodeMixin, Generic[TOPODS]):
    ...
    def __copy__(self) -> Self:
        reference = copy.deepcopy(self)
        if self.wrapped is not None:
            assert (reference.wrapped is not None)  # Ensure mypy knows reference.wrapped is not None
            reference.wrapped.TShape(self.wrapped.TShape())
        return reference
```

Is there a good reason to be doing things this way? I tried replacing it with the following, and so far I'm not seeing any issues:
```python
class Shape(NodeMixin, Generic[TOPODS]):
    ...
    def __copy__(self) -> Self:
        cls = self.__class__
        result = cls.__new__(cls)
        result.__dict__.update(self.__dict__)
        return result
```

Is this a bug for is there some purpose to the deepcopy that I have missed?

I found this behavior when trying to improve the `export_step` to more accurately apply labels to shapes (see issue [NOT SUBMITTED YET] and possibly fix #301 while I'm at it, as well as foundation for #296, where it is important that TShapes are preserved when possible, and the current copy code seems to somehow break that - although I'm unsure exactly why.

jsmnbom, 2025-03-28:
Nevermind my "fix" definitely breaks other things. Back to the drawing board then :/

jsmnbom, 2025-03-28:
Okay so it seems like the cadquery OCP does not expose the proper initializer for TopoDS_Shape to be able to copy it (https://github.com/Open-Cascade-SAS/OCCT/blob/876ccbe977a7a8d4910e68fd6b3536645818ed0e/src/ModelingData/TKBRep/TopoDS/TopoDS_Shape.hxx#L53). We can easily get around that by just using .Located() which calls it for us. So I *think* a proper copy function looks like this:
```
class Shape(NodeMixin, Generic[TOPODS]):
    ...
    def __copy__(self) -> Self:
        cls = self.__class__
        result = cls.__new__(cls)
        for key, value in self.__dict__.items():
            if key == "wrapped":
                result.wrapped = self.wrapped.Located(self.wrapped.Location())
            else:
                setattr(result, key, copy.copy(value))
        return result
```

gumyr, 2025-03-29:
There was no attempt to make the `__copy__` function faster than `__deepcopy__` just to make the use of copied objects faster.  This looks like a promising way to add some performance though. The parent and joint logic of deepcopy would need to be replicated here wouldn't they?