gumyr/build123d #956: ImportError: libvtkWrappingPythonCore3.11-9.3.so: cannot open shared object file: No such file or directory

mainframenzo, 2025-03-30:
Hello!

I am trying to upgrade from 0.6.x to 0.9.x and running into an issue doing so. The issue I see after upgrading is: `ImportError: libvtkWrappingPythonCore3.11-9.3.so: cannot open shared object file: No such file or directory`. I'm using conda to install build123d (in an Ubuntu 22.04 container with a `--platform linux/amd64` flag on a host machine that's an Apple M3). I am probably doing something silly, but I was hoping someone here might have an idea as to what's going on? Thanks!

Here is the original conda yaml file, `build123d-legacy.yaml`:
```
name: build123d-legacy-conda
channels:
  - conda-forge
  - cadquery
  - defaults
dependencies:
  - python=3.11.2
  - cadquery=master
  - pip=24.2
  - pip:
    - build123d==0.6.0
    - cadquery-ocp<7.8
    - cadquery<7.8
    - numpy==1.24.1
    - ocp-tessellate==3.0.10
    - ocp-vscode==2.6.4 
```

Here is the updated conda yaml file, `build123d.yaml`:
```
name: build123d-conda
channels:
  - conda-forge
  - cadquery
  - defaults
dependencies:
  - python=3.11.2
  - cadquery=master
  - pip=24.2
  - pip:
    - build123d==0.9.1
    - cadquery-ocp>=7.8,<7.9
    - cadquery<7.8
    - numpy>=2,<3
    - ocp-tessellate==3.0.10
    - ocp-vscode==2.6.4 
```

```bash
> conda env create --name build123d-legacy-conda --file=build123d-legacy.yaml
> conda env create --name build123d-conda --file=build123d.yaml
> conda activate build123d-legacy-conda
> python3 /opt/app/src/parts-library-tools/test.py
Compound                       at 0x7fff9fbf3470, Center(0.0, 0.0, 0.0)
└── Solid                      at 0x7fff9fbde670, Center(0.0, 0.0, 0.0)
....
> conda deactivate
> conda activate build123d-conda
> python3 /opt/app/src/parts-library-tools/test.py
Traceback (most recent call last):
  File "/opt/app/src/parts-library-tools/test.py", line 3, in <module>
    from build123d import *
  File "/opt/conda/envs/build123d-conda/lib/python3.11/site-packages/build123d/__init__.py", line 3, in <module>
    from build123d.build_common import *
  File "/opt/conda/envs/build123d-conda/lib/python3.11/site-packages/build123d/build_common.py", line 59, in <module>
    from build123d.geometry import (
  File "/opt/conda/envs/build123d-conda/lib/python3.11/site-packages/build123d/geometry.py", line 48, in <module>
    import OCP.TopAbs as TopAbs_ShapeEnum
  File "/opt/conda/envs/build123d-conda/lib/python3.11/site-packages/OCP/__init__.py", line 1, in <module>
    from OCP.OCP import *
ImportError: libvtkWrappingPythonCore3.11-9.3.so: cannot open shared object file: No such file or directory
```

The Python file, `/opt/app/src/parts-library-tools/test.py`:
```
# References:
# * https://build123d.readthedocs.io/en/latest/introductory_examples.html#simple-rectangular-plate
from build123d import *

length, width, thickness = 80.0, 60.0, 10.0

with BuildPart() as ex1:
  Box(length, width, thickness)

print(ex1.part.show_topology())
```

jdegenstein, 2025-04-04:
almost certainly related to https://github.com/gumyr/build123d/issues/871

can you try the solution there and let me know if it helps?

mainframenzo, 2025-09-02:
I am running into the same issue, `ImportError: libvtkWrappingPythonCore3.11-9.3.so: cannot open shared object file: No such file or directory`, using the previous solution! Any ideas on what might have changed?

This was the conda env file that worked previously:
```
name: build123d-conda
channels:
  - conda-forge
  - cadquery
  - defaults
dependencies:
  - python=3.11.2
  - cadquery=master
  - pip=24.2
  - pip:
    - build123d==0.9.1
    - cadquery-ocp>7.8
    - cadquery<7.8
    - numpy>=2,<3
    - ocp-tessellate==3.0.10
    - ocp-vscode==2.6.4
```

bernhard-42, 2025-09-02:
A few hints:
- Do **not** mix conda and pip installations of cadquery and build123d
- You try to install cadquery from conda (*=master*) and then from pip (*<7.8*). The will cause failures
- Try to avoid pinning modules like ocp-tessellate that are anyhow dependencies of ocp_vscode
- No need to add numpy. Will come as a dependency 
- For license reasons try to avoid channel "defaults". It is not needed.

But the biggest issue is: As long as cadquery 2.6 is not on pypi, build123d and cadquery will not work properly together (dependency issues)

My typical approach is:

*build123d.yml:*

```yaml
name: build123d-conda
channels:
  - conda-forge
  - cadquery
dependencies:
  - python=3.11.2
  - pip=25.2
  - pip:
    - build123d==0.9.1
    - ocp-vscode==2.9.0
```

and then use

```bash
conda env create --name build123d-conda --file=build123d.yml
conda activate build123d-conda
```

This properly installs build123d and ocp_vscode with all their dependencies

If you need cadquery, things get a bit less straightforward:
```
conda activate build123d-conda
env -u CONDA_PREFIX pip install --use-pep517 git+https://github.com/cadquery/cadquery
```

`env-u CONDA_PREFIX` : cadquery prevents per default pip installations into a conda environment
`--use-pep517`: cadquery does not use pyproject.toml, but still setup.py

closed 2025-09-02 by mainframenzo