gumyr/build123d #966: Consider adding Color ENUMs or switching to Web Color names

gumyr, 2025-04-18:
The Color names correspond to OCCT defined Enum values.  A Python Enum could be dynamically generated from the OCCT Enum as follows:
```py
from enum import Enum
import OCP.Quantity as Quantity
import inspect

# Use inspect.getmembers to access enum attributes
enum_members = [
    (name.replace("Quantity_NOC_", ""), value)
    for name, value in inspect.getmembers(Quantity.Quantity_NameOfColor)
    if name.startswith("Quantity_NOC_")
]

# Dynamically create the Python Enum
Color = Enum("Color", {name: value for name, value in enum_members})

# Example usage - in geometry.py
color = Color.BLACK
...
self.wrapped = Quantity_Color(color.value)
```

Alternatively, the https://webcolors.readthedocs.io/en/latest/ package provides a simple interface to the Web Colors which could replace the OCCT names entirely.

This issue is related to https://github.com/gumyr/build123d/issues/856

KarthikDani, 2025-04-19:
Amazing

jwagenet, 2025-04-19:
If we want an Enum, but not a massive list, the basic colors could be used: https://en.wikipedia.org/wiki/Web_colors#Basic_colors
However, I think using `webcolors` and the extended color string list is fine here.