gumyr/build123d #971: Build123D Video Tutorials of the Example Designs listed

KarthikDani, 2025-04-20:
Hello, 

I've recently curated a playlist of `build123d` example designs listed on the documentation and shown the development step by step.

This is advantageous in situations where:
1. User needs to understand what part of the code corresponds to what operation in the design. Since as a beginner entire code gives no visualization of the series of operations being performed.
2. One can sit back and watch step-by-step development of design without having to try out the code themselves.

YouTube Playlist can be found through [this link](https://youtube.com/playlist?list=PLzPRaOjzkbhTyW4HbHhf5t9toERC157s8&si=lu3w13ZYN9-kd4L7) 

 Can I PR to add the Playlist in the documentation? As more videos will be put there.

jdegenstein, 2025-04-22:
At the very least you should open a PR on this repo https://github.com/phillipthelen/awesome-build123d possibly under the documentation category. I am not sure where your playlist would be appropriate on the main build123d docs.

KarthikDani, 2025-04-22:
Sure thank you! Will do that soon.