gumyr/build123d #973: normal_at() procudes incorrect values for curved faces, position_at() goes out of bounds of faces

dalibor-frivaldsky, 2025-04-22:
I will try to illustrate the problem in multiple examples below. All the examples have these expectations:

- position_at(u, v) always returns a point that lies on and within the face when **u** ∈ <0, 1> and **v** ∈ <0, 1>
- normal_at(u, v) returns a vector that is normal of the face at the point of position_at(u, v)

This is a helper code used by all the examples:
```
# %%
import numpy as np
from build123d import *
from ocp_vscode import *
from typing import Any, List

mm = 1.0

def visualize_face_normals(
    faces: List[Face],
    num_u_points: int = 10,
    num_v_points: int = 10
) -> List[Any]:
    normals = []
    for face in faces:
        for u in np.linspace(0, 1, num_u_points):
            for v in np.linspace(0, 1, num_v_points):
                point = face.position_at(u, v)
                normal = face.normal_at(u, v)
                with BuildLine() as line:
                    Line(point, point + normal)
                normals.append(line.line)
    return [
        *faces,
        Compound(
            label='normals',
            children=normals
        )
    ]
```

### Example 1 - simple rectagular face
```
# %%
def rectangle_test():
    with BuildSketch() as rect:
        Rectangle(10*mm, 20*mm)

    return rect.faces()

show(
    visualize_face_normals(rectangle_test())
)
```

[image]

Works as expected

### Example 2 - box
```
# %%
def box_test():
    with BuildPart() as box:
        Box(10*mm, 20*mm, 30*mm)

    return box.faces()

show(
    visualize_face_normals(box_test())
)
```

[image]

Works as expected

### Example 3 - simple loft
```
# %%
def loft_test():
    with BuildPart() as test:
        with BuildSketch():
            Rectangle(5*mm, 10*mm)
        with BuildSketch(Pos(0, 0, 20*mm)):
            Rectangle(10*mm, 15*mm)
        loft()

    return test.faces()

show(
    visualize_face_normals(loft_test())
)
```

[image]

This produces unexpected results for the bottom and top faces:
- position_at() returns values out of bounds of the faces
- the u and v coordinate system is rotated about the Z axis

### Example 4 - loft of filleted faces (showing only the problematic faces)
```
# %%
def filleted_loft_test():
    with BuildPart() as test:
        with BuildSketch() as base:
            Rectangle(10*mm, 10*mm)
            fillet(base.vertices(), 2*mm)
        with BuildSketch(Pos(0, 0, 20*mm)) as top:
            Rectangle(10*mm, 10*mm)
            fillet(top.vertices(), 2*mm)
        loft()

    return [test.faces()[i] for i in [1, 3, 5, 7]]
    # return test.faces()

show(
    visualize_face_normals(filleted_loft_test())
)
```

[image]

This procudes unexpected results for the curved corner faces:
- except for the face on the right in the picture (positive X positive Y quadrant), normal_at doesn't produce correct normals
- the normals are progressively more rotated about the Z axis in a direction away from the face's center. The issue is more pronounced for each subsequent face

### Example 5 - straight sweep of a convex face
```
# %%
def sweep_convex_test():
    with BuildPart() as test:
        with BuildSketch(Plane.XZ) as profile:
            with BuildLine():
                l = Polyline(
                    (10*mm, 0),
                    (0, 0),
                    (0, 10*mm)
                )
                RadiusArc(l @ 1, l @ 0, 10*mm)
            make_face()
        with BuildLine() as path:
            Line(
                (0, 0, 0),
                (0, 20*mm, 0)
            )
        sweep(profile.sketch, path)

    return test.faces()

show(
    visualize_face_normals(sweep_convex_test())
)
```

[image]

This produces unexpected results for the side faces:
- position_at() returns values outside of the bounds of the face

### Example 6 - straight sweep of a concave face
```
# %%
def sweep_concave_test():
    with BuildPart() as test:
        with BuildSketch(Plane.XZ) as profile:
            with BuildLine():
                l = Polyline(
                    (10*mm, 0),
                    (0, 0),
                    (0, 10*mm)
                )
                RadiusArc(l @ 1, l @ 0, -10*mm)
            make_face()
        with BuildLine() as path:
            Line(
                (0, 0, 0),
                (0, 20*mm, 0)
            )
        sweep(profile.sketch, path)

    return test.faces()

show(
    visualize_face_normals(sweep_concave_test())
)
```

[image]

This produces unexpected results:
- for the side faces, position_at() returns results outside of the bounds of the face
- for the curved concave surface, the normals are incorrect, rotated about the Y axis in the direction towards the center of the face. This effect is more pronounced towards the edges of the face

### Example 7 - sweep of a convex face along a filleted rectangular path
```
# %%
def sweep_filleted_path_test():
    with BuildPart() as test:
        with BuildSketch(Plane.XZ) as profile:
            with BuildLine():
                l = Polyline(
                    (10*mm, 0),
                    (0, 0),
                    (0, 10*mm)
                )
                RadiusArc(l @ 1, l @ 0, 10*mm)
            make_face()
        with BuildLine() as path:
            FilletPolyline(
                (0, 0),
                (40*mm, 0),
                (40*mm, 40*mm),
                (0, 40*mm),
                radius=15*mm,
                close=True
            )
        sweep(Pos(path.line@0) * Rot(0, 0, 180) * profile.sketch, path)

    return test.faces()

show(
    visualize_face_normals(sweep_filleted_path_test())
)
```

[image]

This produces unexpected results:
- for the bottom flat face, where position_at() returns values outside of the bounds of the face
- the normals of the outer curved corner faces are off, except for the one in +X+Y quadrant
- - the normals are rotated around the Z axis in the direction away from the center of the face. The effect is progressively more pronounced towards the edges of the faces and also for each subsequent face.
- the normals of the inner curved convex corner faces are completely off 
- - seemingly rotated about multiple axes, progressively more towards the edges of the faces and also for each subsequent face

gumyr, 2025-04-22:
The u, v space is defined by the OCCT CAD kernel and it could be larger than the Face and it would also include holes in the Face so "position_at(u, v) always returns a point that lies on and within the face when u ∈ <0, 1> and v ∈ <0, 1>" isn't true.

However, there was a bug in how the normals were calculated that has been fixed.

Thank you for the detailed issue.

dalibor-frivaldsky, 2025-04-22:
Thank you for the quick fix of the normals!

I understand the u, v space, I see I can check for the point being inside the face with Face.is_inside().

What about the rotated u, v space of the loft in example 3? That one I didn't expect to happen.

gumyr, 2025-04-23:
> What about the rotated u, v space of the loft in example 3? That one I didn't expect to happen.

I didn't expect that either but build123d doesn't have any control over the uv space.

closed 2025-04-24 by gumyr