gumyr/build123d #979: Basic full_round returns extra vertex

MatthiasJ1, 2025-04-29:
```py
s = Rectangle(34, 10)
s = full_round((s.edges()<<X)[0])
```

[image]

gumyr, 2025-04-29:
Yes, that's how it's defined:
```py
def full_round(
    edge: Edge,
    invert: bool = False,
    voronoi_point_count: int = 100,
    mode: Mode = Mode.REPLACE,
) -> tuple[Sketch, Vector, float]:
    """Sketch Operation: full_round

    Given an edge from a Face/Sketch, modify the face by replacing the given edge with the
    arc of the Voronoi largest empty circle that will fit within the Face.  This
    "rounds off" the end of the object.

    Args:
        edge (Edge): target Edge to remove
        invert (bool, optional): make the arc concave instead of convex. Defaults to False.
        voronoi_point_count (int, optional): number of points along each edge
            used to create the voronoi vertices as potential locations for the
            center of the largest empty circle. Defaults to 100.
        mode (Mode, optional): combination mode. Defaults to Mode.REPLACE.

    Raises:
        ValueError: Invalid geometry

    Returns:
        (Sketch, Vector, float): A tuple where the first value is the modified shape, the second the
        geometric center of the arc, and the third the radius of the arc

    """
```
You can ignore the other two values by doing:
```py
s,_,_ = full_round((s.edges()<<X)[0])
```
Are you asking that they get removed?

MatthiasJ1, 2025-04-29:
I didn't realize this was expected behavior. IMO this contradicts the rest of the API. All other operations return the new cad object, not cad object + metadata

gumyr, 2025-04-30:
I agree it's not consistent.  One can find the center and radius of the arc with `edge.arc_center` and `edge.radius` so I don't think it's that useful anyway.  I'll just remove them.

closed 2025-05-01 by gumyr