gumyr/build123d #985: Print output from import_svg_as_buildline_code from a shell, build123d.cli:main

westurner, 2025-05-11:
STORY: As a user, I can check the build123d code generated from `import_svg_as_buildline_code("filename.svg")` and iterate in a shell

What would solve:

- [x] A. This one-liner in my dotfiles:
   ```sh
   python -c "import build123d,sys; print(build123d.importers.import_svg_as_buildline_code(sys.argv[1])[0])" filename.svg
   ```
- [ ] B. ENH: `python -m build123d.importers`
  - could expose all/some `import_*` methods and their args to the CLI

- [ ] C. ENH: `python -m build123d`
  - ENH: build123d.cli: argparse

Ways to add CLI scripts:

- ```
  $ touch build123d/__main__.py
  # echo "from build123d.cli import main; main()" > __main__.py
  $ python -m build123d
  ```

- ```toml
  # pyproject.toml
  [project.scripts]
  build123d = build123d.cli:main
  ```

gumyr, 2025-05-11:
This is an interesting idea but I'm not sure I understand how the CLI would work.  As build123d is a CAD library there is no default output from running a script; one could be generating code as you describe, or more commonly creating an object and exporting it in various formats (like SVG, STEP, STL, etc.).  Are you thinking argparse would be used to add arguments like the variable name of the object and a export format like: `build123d --object "my_thing" --export STEP my_file.py`?

westurner, 2025-05-12:
Yeah I have that limited shim there. Should I contribute it? To where, and then what would be in the CLI, and shouldn't there be just one cli:main called by `__main__.py` if there is one at all.

For the so inlclined,

***

Create a CLI with argparse for calling various build123d functions:
- Export
- Import
  - import_svg_as_buildline_code

Add the output of `build123d --help` to the `docs/`, along with a reference and link to argcomplete.

Add support for argcomplete to the argparse parser.