gumyr/build123d #987: Solid.extrude_taper might flip the direction or extrusion

gumyr, 2025-05-12:
This code shows that the `LocOpe_DPrism` call in `Solid.extrude_taper` can invert the direction of extrusion.
```py
from build123d import *
from ocp_vscode import *

from dataclasses import dataclass, field
from typing import Tuple, Optional

@dataclass
class BaseCap:
    width: float
    length: float
    height: float
    wall: float
    roof: float
    taper: float = 0
    stem_rotation: float = 0
    align: Tuple[Align, Align] = (Align.CENTER, Align.CENTER)
    mode: Mode = Mode.ADD

    _part: Optional[Part] = field(init=False, default=None, repr=False)

    left: Optional[Edge] = field(init=False, default=None)
    right: Optional[Edge] = field(init=False, default=None)
    top: Optional[Edge] = field(init=False, default=None)
    bottom: Optional[Edge] = field(init=False, default=None)

    @property
    def part(self) -> Part:
        if self._part is None:
            self._part = self._build().mirror(Plane.XY)
        return self._part

    @part.setter
    def part(self, value: Part) -> None:
        self._part = value

    def _build(self) -> Part:
        with BuildPart() as bp:
            sketch = self._draw()
            print(sketch.face().normal_at())
            extrude(sketch, amount=self.height, taper=self.taper)
        return bp.part

    def _draw(self) -> Sketch:
        raise NotImplementedError

@dataclass
class AsymmetricCap(BaseCap):
    angle: float = 0
    flip: bool = False

    def _draw(self) -> Sketch:
        d = -1 if self.flip else 1

        with BuildSketch() as sk:
            with BuildLine():
                self.right = Line((0, 0), (0, -self.length))
                self.bottom = PolarLine(
                    start=self.right @ 1,
                    length=self.width * d,
                    angle=180 + self.angle * d,
                )
                self.left = PolarLine(
                    start=self.bottom @ 1,
                    length=(self.right @ 0).Y - (self.bottom @ 1).Y,
                    angle=90 + self.angle * d,
                    length_mode=LengthMode.VERTICAL,
                )
                self.top = Line(self.left @ 1, self.right @ 0)
            make_face()

        if self.flip:
            self.left, self.right = self.right, self.left

        return sk.sketch

c1 = AsymmetricCap(
    width=10,
    length=20,
    height=5,
    wall=2,
    roof=1,
    angle=4,
    taper=1,
)

c2 = AsymmetricCap(
    width=10,
    length=20,
    height=5,
    wall=2,
    roof=1,
    angle=8,
    taper=0,
    flip=True,
)

c2.part.move(Pos(10))

show(c1.part, c2.part)
```

jdegenstein, 2026-10-07:
@gumyr can this be closed too?

gumyr, 2026-10-07:
Yes

closed 2026-10-07 by gumyr