gumyr/build123d #990: Offset on a face creates an impossible shape

blastrock, 2025-05-14:
Hi!

The following script produces an impossible shape. It's supposed to produce an offset face, but it makes a line with weird a loop:

```py
from build123d import (
    Line,
    BuildLine,
    BuildSketch,
    add,
    make_face,
    offset,
)

with BuildLine() as edge_cuts:
    Line(
        (142.5, 12.8),
        (-8.0, -94),
    )
    Line(
        (-8.0, -94),
        (72.771610, -94),
    )
    Line(
        (72.771610, -94),
        (73.373425, -94.201365),
    )
    Line((73.373425, -94.201364), (120.733112, -129.889448))
    Line(
        (120.733112, -129.889448),
        (142.5, 12.8),
    )

with BuildSketch() as bottom_sketch:
    add(edge_cuts)
    make_face()
    offset(amount=0.5)

show_object(bottom_sketch)
```

It produces the following shape:

[image]

[image]

If I comment the `offset` call, I get the following:

[image]

Version: cq-editor 0.30-dev-8
According to the release note:
Based on CadQuery/cadquery from Feb 20, 2025
Based on gumyr/build123d from Feb 20, 2025 (build123d 0.9.1+)

blastrock, 2025-05-14:
Wow, good catch! Thanks a lot! The source is a PCB file from kicad, and my script actually loads it dynamically. I reduced the example to multiple Line calls manually.

My first thought was that the line must be open somewhere, but I thought that if a face could be produced without the offset, it meant that the line must be closed. Is there a good way to check whether a line is correctly closed?

gumyr, 2025-05-14:
Without the fix:
```py
print(edge_cuts.wire().is_closed)
```
```
False
```

blastrock, 2025-05-15:
Awesome, thank you!!

closed 2025-05-15 by blastrock