gumyr/build123d #995: Bug: Compound export to gltf does not capture colors (but 3mf export does)

MRiabov, 2025-05-23:
Hello,
I am doing something that needs colored gltfs (namely, I want to import the colored data to open3d)

So, I'm doing:
part creation:
```python
import build123d as bd
import ocp_vscode

with bd.BuildPart() as test:
    bd.Box(10, 10, 10)

with bd.BuildPart() as test2:
    with bd.Locations((15, 0, 0)):
        bd.Box(10, 10, 10)

test = test.solid()
test2 = test2.solid()
test.label = "test"
test.color = bd.Color("blue")
test2.color = bd.Color("red")

test_comp = bd.Compound((test, test2))

# export to gltf:
bd.export_gltf(test_comp, "test.gltf")

#export to 3mf:
mesher = bd.Mesher()
mesher.add_shape([test, test2])
mesher.write("test.3mf")

#view:
ocp_vscode.show(test_comp)
```

Here, ocp_vscode will capture no colors, 3mf will capture colors, bd.export_gltf won't.

Test:
Go to: https://3dviewer.net/, and import the 3mf file, the colors will show:

[image]

Gltf won't:

[image]

STL also won't, nor will glb.

Linked: #929 

Please fix. This blocks me.

MRiabov, 2025-05-23:
Edit: setting `ocp_vscode.show([test, test2])`(parts, not compound) allows to bypass the compound and render ocp_vscode with color. Nevertheless, it shouldn't be the case that I should manually paste 20 parts in there, when I have 20 parts

gumyr, 2025-05-23:
There isn't a problem with color export to glb, the problem is in how you created your `Compound` - a `Shape` and not an assembly. A new `Assembly` class will be introduced to help clarify the difference but for now see this:
```py
from build123d import *
import ocp_vscode

with BuildPart() as test:
    Box(10 * M, 10 * M, 10 * M)

with BuildPart() as test2:
    with Locations((15 * M, 0, 0)):
        Box(10 * M, 10 * M, 10 * M)

test = test.solid()
test2 = test2.solid()
test.label = "test"
test.color = Color("blue")
test2.color = Color("red")

# test_comp = Compound((test, test2))
test_assembly = Compound(children=[test, test2])

# export to gltf:
# export_gltf(test_comp, "test.gltf")
export_gltf(test_assembly, "test.gltf")
```

[image]

The key difference is the use of `children` here: `Compound(children=[test, test2])` which tells build123d that you want an assembly and not just an OpenCascade `Compound` object.  The documentation describes this here: https://build123d.readthedocs.io/en/latest/assemblies.html.  I've also changed the size of the parts as the default unit for blender is a meter.

There is a an existing bug for the 3MF export of color: https://github.com/gumyr/build123d/issues/929

MRiabov, 2025-05-23:
Hmm, OK. I didn't know there is a difference between Compound((**objs)) and Compound(children=[]). Well, this explains things.

Thanks.

closed 2025-05-23 by gumyr