gumyr/build123d #1004: Wrap: "IndexError: list index out of range" when it seemingly should be a threshold too low error.

seltzered, 2025-05-31:
In testing the new `wrap` function, I noticed something where I would expect an error about my tolerance being too low but instead get a much lengthier traceback ending in `IndexError: list index out of range`. 

The example test case I have for this is an attempt to wrap a triangle onto a cylinder, where the code runs fine if the wrap threshold is set to 1.9 or higher, but the error happens with the default threshold or any threshold at 1.8 or below:

```
# [Imports]
from build123d import *
from ocp_vscode import *

import math

with BuildPart() as cylinder_with_wrap:

    cylinder_radius = 29.2 * MM
    cylinder_height = 73.2 * MM
    my_cylinder = Cylinder(radius=cylinder_radius,
                          height=cylinder_height,
                          rotation=(0,0,180))

    param_radius = 19.0 * MM
    
    with BuildSketch() as towrap_sketch:
        with BuildLine() as towrap_line:

            l1 = Line((param_radius * 0.5, param_radius), 
                        (param_radius * 0.5, -param_radius))

            l2 = Line(l1 @ 1, 
                      ((param_radius * 1.5) + param_radius,
                       -param_radius))

            l3 = Line(l2 @ 1,
                      l1 @ 0)

        make_face()

    my_cylinder_face = my_cylinder.faces().filter_by(GeomType.CYLINDER)[0]
    towrap_face = towrap_sketch.face()

    target = my_cylinder_face.location_at(
        u=0.5, 
        v=0.5,
        x_dir=(1.0, 0.0, 0.0))
    towrap_wrap = my_cylinder_face.wrap(
        planar_shape=towrap_face,
        surface_loc=target,
        tolerance=1.8            # Change to 1.9 or higher to avoid error
        )
    
show(my_cylinder, towrap_wrap, towrap_sketch, target) 
```

The traceback that occurs is 
```
Traceback (most recent call last):
  File "/home/me/build123d_wrap_cylinder_demo.py", line 41, in <module>
    towrap_wrap = my_cylinder_face.wrap(
        planar_shape=towrap_face,
        surface_loc=target,
        tolerance=1.8
        )
  File "/home/me/.pyenv/versions/3.13.3/lib/python3.13/site-packages/build123d/topology/two_d.py", line 1766, in wrap
    return self._wrap_face(
           ~~~~~~~~~~~~~~~^
        planar_shape, surface_loc, tolerance, extension_factor
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/home/me/.pyenv/versions/3.13.3/lib/python3.13/site-packages/build123d/topology/two_d.py", line 1865, in _wrap_face
    wrapped_perimeter = self._wrap_wire(
        planar_face.outer_wire(), surface_loc, tolerance, extension_factor
    )
  File "/home/me/.pyenv/versions/3.13.3/lib/python3.13/site-packages/build123d/topology/two_d.py", line 1956, in _wrap_wire
    wrapped_edge: Edge = self._wrap_edge(
                         ~~~~~~~~~~~~~~~^
        local_planar_edge, edge_surface_location, True, tolerance
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/home/me/.pyenv/versions/3.13.3/lib/python3.13/site-packages/build123d/topology/two_d.py", line 368, in _wrap_edge
    current_point, current_normal = _find_point_on_surface(
                                    ~~~~~~~~~~~~~~~~~~~~~~^
        current_point, current_normal, offset
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/home/me/.pyenv/versions/3.13.3/lib/python3.13/site-packages/build123d/topology/two_d.py", line 324, in _find_point_on_surface
    return _intersect_surface_normal(
        world_point, world_point - target_object_center
    )
  File "/home/me/.pyenv/versions/3.13.3/lib/python3.13/site-packages/build123d/topology/two_d.py", line 304, in _intersect_surface_normal
    face = self.faces_intersected_by_axis(axis).sort_by(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        lambda f: f.distance_to(point)
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    )[0]
    ~^^^
  File "/home/me/.pyenv/versions/3.13.3/lib/python3.13/site-packages/build123d/topology/shape_core.py", line 2372, in __getitem__
    return list(self).__getitem__(key)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^
IndexError: list index out of range
```
```

seltzered, 2025-05-31:
Thanks, I have one more test case that might fall under scenario 2 as you mention to the point where a closed face ends up seemingly splitting out. Could also be relevant for #1002 . 

```
# [Imports]
from build123d import *
from ocp_vscode import *

import math

with BuildPart() as cylinder_with_wrap:

    cylinder_radius = 29.2 * MM
    cylinder_height = 73.2 * MM
    my_cylinder = Cylinder(radius=cylinder_radius,
                          height=cylinder_height,
                          rotation=(0,0,180))

    param_radius = 19.0 * MM
    
    with BuildSketch() as towrap_sketch:
        with BuildLine() as towrap_line:

            l1 = Line((param_radius * 0.5, param_radius), 
                      (0.0, param_radius))
            l2 = JernArc(l1 @ 1, l1 % 1, param_radius, 180)
            l3 = Line(l2 @ 1, 
                      ((param_radius * 1.5) + param_radius,
                        -param_radius))
            l4 = JernArc(l3 @ 1, l3 % 1, 
                         param_radius, 90)
            l5 = Line(l4 @ 1, 
                      (((param_radius * 1.5) + param_radius) + param_radius,
                       param_radius ))
            l6 = Line(l5 @ 1, 
                      l1 @ 0)

        make_face()

    my_cylinder_face = my_cylinder.faces().filter_by(GeomType.CYLINDER)[0]
    towrap_face = towrap_sketch.face()

    target = my_cylinder_face.location_at(
        u=0.4, 
        v=0.7,
        x_dir=(1.0, 0.0, 0.0))
    towrap_wrap = my_cylinder_face.wrap(
        planar_shape=towrap_face,
        surface_loc=target,
        tolerance=6.0
        )
    
show(my_cylinder, towrap_wrap, towrap_sketch, target) 
```

gumyr, 2026-09-22:
Fixed with PR #1465. Tolerance no longer controls sampling, so the failure mode disappears.

closed 2026-09-22 by gumyr