gumyr/build123d #1011: OCCT fillet is operating on all of the edges in the wire not just the one selected

gumyr, 2025-06-15:
```py
with BuildPart() as phone_case:
    with BuildSketch():
        RectangleRounded(80 * MM, 150 * MM, 10 * MM)
        with Locations((-20 * MM, 55 * MM)):
            RectangleRounded(25 * MM, 25 * MM, 2 * MM, mode=Mode.SUBTRACT)
    extrude(amount=2 * MM)
    to_fillet = phone_case.faces().sort_by(Axis.Z)[-1].inner_wires()[0].edges()[0]
    # Although only a single edge is selected - and corrected sent to OCCT - all the
    # edges in the inner wire are filleted
    fillet(to_fillet, 1)
```

[image]

MatthiasJ1, 2025-06-15:
Does OCCT expose such functionality?

gumyr, 2025-06-16:
I don't understand your question, this problem is with the standard `BRepFilletAPI_MakeFillet`.  Only one `Edge` gets added but multiple get filleted.

MatthiasJ1, 2025-06-16:
What I mean is it possible to make BRepFilletAPI_MakeFillet not do that.

jdegenstein, 2025-06-16:
I agree with @MatthiasJ1 -- typically edges that are connected with C1+ continuity are automatically filleted together. I actually use this frequently while modeling to fillet a continuous edge chain that might be difficult to isolate.

gumyr, 2025-06-16:
> What I mean is it possible to make BRepFilletAPI_MakeFillet not do that.

Maybe, but it will need to be investigated.

gumyr, 2025-07-08:
There doesn't appear to be any mechanism to control this behaviour in `BRepFilletAPI_MakeFillet` but hopefully I'm wrong.