gumyr/build123d #1012: Plane.XZ top split seems reversed

gautaz, 2025-06-16:
Hello,

The following code:
```Python
with BuildPart():
    Sphere(30)
    split(bisect_by=Plane.XY, keep=Keep.TOP)
    split(bisect_by=Plane.XZ, keep=Keep.TOP)
    split(bisect_by=Plane.YZ, keep=Keep.TOP)
```

Results in:

[image]

So the top parts seem to have been all preserved but for the Plane.XZ split.

The 3D coordinate system displayed in my viewer seems OK, what am I missing?

jdegenstein, 2025-06-16:
All coordinate systems in build123d use the right hand rule:
```
Plane.XY has a normal in the +Z direction
Plane.YZ has a normal in the +X direction
Plane.XZ has a normal in the -Y direction!
```
https://en.wikipedia.org/wiki/Right-hand_rule

gautaz, 2025-06-16:
Ah, ok, I should have used the Plane.ZX.
I assumed only Plane.XY, Plane.YZ and Plane.XZ existed and did not realize that the axis were intentionally ordered in these names...
Thanks @jdegenstein for the clarification.

closed 2025-06-16 by gautaz