gumyr/build123d #1027: Pos() objects can't be pre-computed, fail unless constructed in the same expression as a '*' operator

andyross, 2025-07-07:
Ran into this bit of weirdness.  I want a big array of Location objects from which to construct copies.  The obvious way is to pre-construct all the Pos() objects in a loop.  But that products a OCP failure.  Easy exerciser:

```python
# Works
b0 = Pos(1, 0, 0) * Box(1, 1, 1)

# Fails
loc_array = [Pos(0, 0, 1)]
b1 = loc_array[0] * Box(1, 1, 1)

# Weird workaround:
loc_tuple_array = [(0, 1, 0)]
b2 = Pos(*(loc_tuple_array[0])) * Box(1, 1, 1)
```

The second box (b1), if left uncommented, produces this failure:

```
  File "/home/andy/b123d-test.py", line 65, in <module>
    yacv_server.show_all(texture=tex)
    ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^
  File "/home/andy/py3d/py/lib/python3.13/site-packages/yacv_server/yacv.py", line 234, in show_cad_all
    all_cad = list(grab_all_cad())  # List for reproducible iteration order
                   ~~~~~~~~~~~~^^
  File "/home/andy/py3d/py/lib/python3.13/site-packages/yacv_server/cad.py", line 99, in grab_all_cad
    shape = get_shape(value, error=False)
  File "/home/andy/py3d/py/lib/python3.13/site-packages/yacv_server/cad.py", line 81, in get_shape
    shapes_bd = [Compound(shape) for shape in shapes_raw_filtered_sorted if shape is not None]
                 ~~~~~~~~^^^^^^^
  File "/home/andy/py3d/py/lib/python3.13/site-packages/build123d/topology/composite.py", line 177, in __init__
    super().__init__(
    ~~~~~~~~~~~~~~~~^
        obj=topods_compound,
        ^^^^^^^^^^^^^^^^^^^^
    ...<2 lines>...
        parent=parent,
        ^^^^^^^^^^^^^^
    )
    ^
  File "/home/andy/py3d/py/lib/python3.13/site-packages/build123d/topology/shape_core.py", line 288, in __init__
    tcast(Optional[TOPODS], downcast(obj)) if obj is not None else None
                            ~~~~~~~~^^^^^
  File "/home/andy/py3d/py/lib/python3.13/site-packages/build123d/topology/shape_core.py", line 3012, in downcast
    f_downcast: Any = Shape.downcast_LUT[shapetype(obj)]
                                         ~~~~~~~~~^^^^^
  File "/home/andy/py3d/py/lib/python3.13/site-packages/build123d/topology/shape_core.py", line 3082, in shapetype
    if obj is None or obj.IsNull():
                      ^^^^^^^^^^
AttributeError: 'OCP.OCP.TopLoc.TopLoc_Location' object has no attribute 'IsNull'
```

andyross, 2025-07-07:
FWIW, it occurs to me that this might be specific to yacv_server (you can see it at the top of the stack above).  Maybe it's trying to walk the graph in show() in a way that the built-in exporters and ocp_vscode don't?

gumyr, 2025-07-08:
I don't see this problem with the build123d/ocp_vscode so I agree it appears to be specific to yacv_server.

closed 2025-07-11 by andyross