gumyr/build123d #1030: Compound construction is superlinear, extremely slow at scale

andyross, 2025-07-11:
The Compound class has what appears to be a O(N^2) performance bug in initialization.  The code below creates a large 3D array of disjoint cubes, then assembles them into a Compound object.  The object creation isn't exactly fast (400us per cube on my device), but it's linear and tolerable.  The Compound() constructor rapidly gets out of hand, however.

At small-integer counts, it's noise.  By a 10x10x10 array, it's taking as long to create the Compound as it did to create the geometry, and at 27x27x27 it looks like this:

```
$ ./compound-perf.py
generated cell 0.0
generated locs 0.0
generated cells 8.1
generated Compound 97.6
exported 0.6
```

Compound shouldn't be doing any processing beyond grouping, isn't that the whole point?  I'm wanting to use this to manage space-filling tesselations like 3D infill and support, and frankly it's just not going to work.

Digging in a bit, it looks like the time is all taken in the assignment to `self.children` in the __init__ method, which implies there's something in OCP/OCCT at fault or interacting, or plausibly the anytree code I see is tripping over a linear step it doesn't expect.

```python
from build123d import *
import time
import itertools
import copy

t_last = time.time()
def chktime(msg):
    global t_last
    t = time.time()
    print(f"{msg} {t-t_last:.1f}")
    t_last = t

cell = Box(1, 1, 1)
bb = cell.bounding_box().size
chktime("generated cell")
latlocs = (Pos(bb.X * 1.1 * x, bb.Y * 1.1 * y, bb.Z * 1.1 * z) for x, y, z
           in itertools.product(range(27), range(27), range(27)))
chktime("generated locs")
latcells = [ l * copy.copy(cell) for l in latlocs ]
chktime("generated cells")
network = Compound(children=latcells)
chktime("generated Compound")
export_stl(network, "tmp.stl")
chktime("exported")
```

jdegenstein, 2025-07-11:
Thanks for reporting, I did a bit more profiling with the following script based on yours above:
```py
from build123d import *
import time
import itertools
import copy
import cProfile
import pstats
rxyz = 20
t_last = time.time()
def chktime(msg):
    global t_last
    t = time.time()
    print(f"{msg} {t-t_last:.1f}")
    t_last = t

cell = Box(1, 1, 1)
bb = cell.bounding_box().size
chktime("generated cell")
latlocs = (Pos(bb.X * 1.1 * x, bb.Y * 1.1 * y, bb.Z * 1.1 * z) for x, y, z
           in itertools.product(range(rxyz), range(rxyz), range(rxyz)))
chktime("generated locs")
latcells = [ l * copy.copy(cell) for l in latlocs ]
chktime("generated cells")
profiler = cProfile.Profile()
profiler.enable()
network=Compound(children=latcells)
profiler.disable()
chktime("generated Compound")

stats = pstats.Stats(profiler)
stats.sort_stats('time').print_stats(20) # Print top 10 time-consuming calls
```
Returns:
```py
         32268079 function calls in 37.876 seconds

   Ordered by: internal time
   List reduced from 58 to 20 due to restriction <20>

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
 32012000   27.021    0.000   27.021    0.000 {built-in method OCP.OCP.TopoDS.Add}
     8001    6.289    0.001   33.329    0.004 venv\Lib\site-packages\build123d\topology\utils.py:201(_make_topods_compound_from_shapes)
     8000    2.176    0.000    2.176    0.000 venv\Lib\site-packages\build123d\topology\composite.py:739(<listcomp>)
     8000    2.042    0.000   37.687    0.005 venv\Lib\site-packages\build123d\topology\composite.py:735(_post_attach)
     8004    0.127    0.000    0.130    0.000 venv\Lib\site-packages\anytree\node\nodemixin.py:178(children)
     8001    0.042    0.000   37.840    0.005 venv\Lib\site-packages\anytree\node\nodemixin.py:123(parent)
     8000    0.030    0.000   37.732    0.005 venv\Lib\site-packages\anytree\node\nodemixin.py:159(__attach)
     8000    0.022    0.000    0.054    0.000 venv\Lib\site-packages\anytree\node\nodemixin.py:137(__check_loop)
     8001    0.019    0.000    0.019    0.000 {built-in method OCP.OCP.TopoDS.MakeCompound}
        1    0.017    0.017   37.876   37.876 venv\Lib\site-packages\anytree\node\nodemixin.py:243(children)
     8001    0.010    0.000    0.018    0.000 venv\Lib\logging\__init__.py:1467(debug)
    16000    0.010    0.000    0.016    0.000 venv\Lib\site-packages\anytree\node\nodemixin.py:303(iter_path_reverse)
     8000    0.010    0.000    0.032    0.000 {built-in method builtins.any}
     8001    0.008    0.000    0.008    0.000 venv\Lib\logging\__init__.py:1734(isEnabledFor)
    32001    0.008    0.000    0.008    0.000 {built-in method builtins.isinstance}
    32005    0.007    0.000    0.007    0.000 {built-in method builtins.hasattr}
    16000    0.007    0.000    0.022    0.000 venv\Lib\site-packages\anytree\node\nodemixin.py:142(<genexpr>)
     8000    0.006    0.000    0.007    0.000 venv\Lib\site-packages\build123d\topology\composite.py:774(_pre_attach)
    16004    0.006    0.000    0.008    0.000 venv\Lib\site-packages\anytree\node\nodemixin.py:172(__children_or_empty)
        1    0.004    0.004    0.007    0.007 venv\Lib\site-packages\anytree\node\nodemixin.py:229(__check_children)
```
This raises several questions for me:
```
1. why is OCP Add method called ~32 million times when 20^3 is only 8,000 boxes. By my count a single cube is composed of 8 vertices, 6 faces, 12 edges, and 1 solid (total of 27 "subshapes") for a total of 216,000 total.
2. can `_make_topods_compound_from_shapes` be optimized to pass the iterable to OCP somehow? Quick investigation says no, but hope I am wrong.

andyross, 2025-07-11:
Nice.

> why is OCP...Add method called ~32 million times when 20^3 is only 8,000 boxes

Because `sum(range(int(math.pow(20, 3)))) == 31996000`

You found the smoking gun, that's the O(N^2) dependency right there.  Something is re-adding the whole set from scratch every time one element is added.  I don't remotely know enough about OCCT to know where to start looking, but those numbers don't lie: that's 100% what is happening somewhere.

andyross, 2025-07-12:
Came back to this a bit this morning.  Before it had looked like it was the set to the children field of the Compound object at fault.  So I added the same timing rig to the Compound class (diff below) and now it looks like this:

```
$ ~/b3d/compound-perf.py
setting children 0.3
ENTER _post_attach_children() 0.0
EXIT _post_attach_children() 0.0
children set 0.0
setting children 0.0
ENTER _post_attach_children() 0.0
EXIT _post_attach_children() 0.0
children set 0.0
setting children 3.3
ENTER _post_attach_children() 18.1
EXIT _post_attach_children() 0.0
children set 0.0
```

Very oddly, the time taken is **not** in the call to `_make_topods_compound_from_shapes()`, it's before it, somewhere between the assignment to `self.children` (to the list passed as an argument from the app) and the entrance to `Compound._post_attach_children`.

I'm a little stumped here, but I think my gut from earlier was right and that there's some kind of dance happening in the anytree node management, where maybe a child gets initialized and tries to normalize its parent or something?

```diff
diff --git a/src/build123d/topology/composite.py b/src/build123d/topology/composite.py
index 63bdb14..ebbf1ca 100644
--- a/src/build123d/topology/composite.py
+++ b/src/build123d/topology/composite.py
@@ -51,7 +51,6 @@ license:
     limitations under the License.

 """
-
 from __future__ import annotations

 import copy
@@ -129,6 +128,13 @@ from .utils import (
 )
 from .zero_d import Vertex

+import time
+t_last = time.time()
+def chktime(msg):
+    global t_last
+    t = time.time()
+    print(f"{msg} {t-t_last:.1f}")
+    t_last = t

 class Compound(Mixin3D, Shape[TopoDS_Compound]):
     """A Compound in build123d is a topological entity representing a collection of
@@ -182,7 +188,9 @@ class Compound(Mixin3D, Shape[TopoDS_Compound]):
         )
         self.material = "" if material is None else material
         self.joints = {} if joints is None else joints
+        chktime("setting children")
         self.children = [] if children is None else children
+        chktime("children set")

     # ---- Properties ----

@@ -741,6 +749,7 @@ class Compound(Mixin3D, Shape[TopoDS_Compound]):

     def _post_attach_children(self, children: Iterable[Shape]):
         """Method call after attaching `children`."""
+        chktime("ENTER _post_attach_children()")
         if children:
             kids = ",".join([child.label for child in children])
             logger.debug("Adding children %s to %s", kids, self.label)
@@ -749,6 +758,7 @@ class Compound(Mixin3D, Shape[TopoDS_Compound]):
             )
         # else:
         #     logger.debug("Adding no children to %s", self.label)
+        chktime("EXIT _post_attach_children()")

     def _post_detach(self, parent: Compound):
         """Method call after detaching from `parent`."""
```

jdegenstein, 2025-11-23:
FYI, I have a branch that passes all tests with changes to the use of anytree here:
https://github.com/jdegenstein/build123d/tree/anytree_suppress

The approach was indeed to bypass issues with hash collision in anytree that I think was causing it to check all new children to be checked against all previous children (resulting in O(n^2) behavior).

I introduced a new test to the benchmarks workflow (`test_compound_many_children`) that on my machine saw the time cut by a factor of 1/3rd (approx. 800ms vs. approx 2700ms). The test currently constructs a compound of 1,728 children (12^3) but more importantly than this single test -- it scales O(n) instead of O(n^2).

Example benchmark run on the aforementioned "fixed" branch:
https://github.com/jdegenstein/build123d/actions/runs/19598835584/job/56127325835