gumyr/build123d #1093: BoundBox.is_inside returns the opposite

nikandfor, 2025-09-21:
```
# for two non-intersecting boxes
b1 = Location((2, 0, 0)) * Box(1, 1, 1)
b2 = Location((-2, 0, 0)) * Box(1, 1, 1)

# it returns True
b1.bounding_box().is_inside(b2.bounding_box())
```

Is [this](https://github.com/gumyr/build123d/blob/790f0eacedcf056cfd46ab9c3945a5a405e6e2e7/src/build123d/geometry.py#L1127) `not` supposed to be here?
```
    def is_inside(self, second_box: BoundBox) -> bool:
        return not (
            second_box.min.X > self.min.X
            and second_box.min.Y > self.min.Y
            and second_box.min.Z > self.min.Z
            and second_box.max.X < self.max.X
            and second_box.max.Y < self.max.Y
            and second_box.max.Z < self.max.Z
        )
```

jdegenstein, 2025-09-22:
Most likely a duplicate of https://github.com/gumyr/build123d/issues/827 and see also this draft PR https://github.com/gumyr/build123d/pull/828

gumyr, 2026-08-20:
Closed with PR #828

closed 2026-08-20 by gumyr