gumyr/build123d #1094: Wire.locations is very slow for complex wire geometries and significant numbers of points

pythoro, 2025-09-22:
I have a complex 3D Wire with about 1500 edges, and I need to compute thousands of points along this wire. I generated a numpy array from 0 to 1 with 2000 divisions, then called wire.locations(arr). It works, but it is surprisingly slow - so slow it makes this nearly too frustrating to use.

gumyr, 2025-09-22:
The `locations` method doesn't assume anything about the locations which is probably not what most users are doing with it - they are looking for locations uniformly distributed along the Edge/Wire.  Here is a prototype of that method which finds 2000 locations on a wire of 1500 edges in 0.175s:

```py
from OCP.GCPnts import GCPnts_UniformAbscissa
from OCP.GeomFill import (
    GeomFill_CorrectedFrenet,
    GeomFill_Frenet,
    GeomFill_TrihedronLaw,
)
from OCP.gp import gp_Ax3, gp_Dir, gp_Trsf, gp_Vec
from OCP.Standard import Standard_ConstructionError
from OCP.TopLoc import TopLoc_Location
from ocp_vscode import show, show_all

from build123d import *
from build123d.topology import Mixin1D

import timeit

def uniform_locations(
    self: Edge | Wire,
    location_count: int,
    frame_method: FrameMethod = FrameMethod.FRENET,
    x_dir: VectorLike | None = None,
) -> list[Location]:

    curve = self.geom_adaptor()
    location_count += int(self.is_closed)
    uniform_absicissa = GCPnts_UniformAbscissa(curve, location_count)
    law: GeomFill_TrihedronLaw
    if frame_method == FrameMethod.FRENET:
        law = GeomFill_Frenet()
    else:
        law = GeomFill_CorrectedFrenet()

    law.SetCurve(curve)
    x_dir_dir = Vector(x_dir).to_dir() if x_dir is not None else None

    location_list = []
    for i in range(uniform_absicissa.NbPoints()):
        param = uniform_absicissa.Parameter(i + 1)
        tangent, normal, binormal = gp_Vec(), gp_Vec(), gp_Vec()

        law.D0(param, tangent, normal, binormal)
        pnt = curve.Value(param)

        transformation = gp_Trsf()
        if x_dir is not None:
            try:

                transformation.SetTransformation(
                    gp_Ax3(pnt, gp_Dir(tangent.XYZ()), x_dir_dir, gp_Ax3())
                )
            except Standard_ConstructionError as exc:
                raise ValueError(
                    f"Unable to create location with given x_dir {x_dir}. "
                    f"x_dir must be perpendicular to shape's tangent "
                    f"{tuple(Vector(tangent))}."
                ) from exc

        else:
            transformation.SetTransformation(
                gp_Ax3(pnt, gp_Dir(tangent.XYZ()), gp_Dir(normal.XYZ())), gp_Ax3()
            )
        loc = Location(TopLoc_Location(transformation))

        if self.is_forward:
            location_list.append(loc)
        else:
            location_list.append(-loc)

    if self.is_closed:
        location_list.pop()
    return ShapeList(location_list)

Mixin1D.uniform_locations = uniform_locations
edge_count = 1500
edge_list = [
    CenterArc((0, 0), 1, 360 * i / edge_count, 360 / edge_count)
    for i in range(edge_count)
]
segmented_wire = Wire(edge_list)

start_time = timeit.default_timer()
locs = segmented_wire.uniform_locations(2000)
print(f"Time: {timeit.default_timer() - start_time:0.3f}s")
pnts = [l.position for l in locs]
b = locs[1000] * Box(0.1, 0.2, 0.3)  # visual check
show(segmented_wire, pnts, b)
```

Maybe `locations` should be replaced by this version?

closed 2026-10-25 by gumyr

gumyr, 2026-10-25:
`locations` (just like `positions`) now has a `deflection` parameter will quickly generates a minimal list of locations that are required to generate the curve to that deflection.